PrusaOwners/OctoPrint-PrusaMeshMap #8: Try to match heatmap scale to bed dimensions

koenkooi, 2018-04-02:
The heatmap scale currently covers approximately -1 to +7 on x and y, while the print area is  approximately 20 x 25cm. 

[image]

An overlay showing the bed outline and/or screw locations would be even better :)

ff8jake, 2018-04-02:
I think we can adjust the size ratio using matplotlib, and could probably use matplotlib to put points on there as well for screws. Probably need to get a concrete answer from Prusa folks on the G81 output though. They return a 7x7 matrix, but we know the bed is a 3x3 matrix. The above looks smooth because of the spline16 filter, here's what the unfiltered output looks like:

[image]

I assume in the above 7x7 grid, that the screws will be squares will be the centers and corners respectively; however, aren't the mesh leveling points offset from these? Just curious if we put the screw dots on there, where exactly we need to put them to not convey false information to the end user. Maybe we should put both screws and sensor points? :)

koenkooi, 2018-04-02:
A quick look at the firmware shows https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/mesh_bed_leveling.cpp#L100

And a comment in main.cpp saying ```bilinear interpolation from 3x3 to 7x7 points while using the same array z_values[iy][ix] for storing (just coppying measured data to new destination and interpolating between them)```

koenkooi, 2018-04-02:
And I think https://github.com/prusa3d/Heatbed_MK52_magnetic/blob/master/24V_05d/Heatbed-MK52.png shows the locations of the probe points

koenkooi, 2018-04-03:
Looking at the 3x3 interpolation some more, the 3x3 matrix
```
a1 a2 a3
b1 b2 b3
c1 c2 c3
```

becomes
```
a1 x x a2 x x a3
x  x x x  x x x
x  x x x  x x x
b1 x x b2 x x b3
x  x x x  x x x
x  x x x  x x x
c1 x x c2 x x c3
```

And the points are defined as following
```
// Positions of the bed reference points in the machine coordinates, referenced to the P.I.N.D.A sensor.
// The points are ordered in a zig-zag fashion to speed up the calibration.
const float bed_ref_points[] PROGMEM = {
    13.f  - BED_ZERO_REF_X,   6.4f - BED_ZERO_REF_Y,
    115.f - BED_ZERO_REF_X,   6.4f - BED_ZERO_REF_Y,
    216.f - BED_ZERO_REF_X,   6.4f - BED_ZERO_REF_Y,

    216.f - BED_ZERO_REF_X, 104.4f - BED_ZERO_REF_Y,
    115.f - BED_ZERO_REF_X, 104.4f - BED_ZERO_REF_Y,
    13.f  - BED_ZERO_REF_X, 104.4f - BED_ZERO_REF_Y,

    13.f  - BED_ZERO_REF_X, 202.4f - BED_ZERO_REF_Y,
    115.f - BED_ZERO_REF_X, 202.4f - BED_ZERO_REF_Y,
    216.f - BED_ZERO_REF_X, 202.4f - BED_ZERO_REF_Y
};
```

koenkooi, 2018-04-03:
Doing a G80 with a hot extruder will leave small PLA blobs at the probe points. Using those and a ruler,  the locations match the grid coordinates printed on the bed, which matches the one printed on the steel sheet.

So
```
a1 a2 a3
b1 b2 b3
c1 c2 c4
```
maps to
```
(13,6.4)    (115,6.4)    (216,6.4)
(13,104.4)  (115,104.4)  (216,104.4)
(13,204.4)  (115,204.4)  (216,204.4)
```

koenkooi, 2018-04-03:
Matplotlib sadly doesn't support image operations nor scaling of heatmaps, so image magick to the rescue:

```
#!/bin/bash

convert heatmap.png -crop 368x360+108+59 map.png
convert map.png  -resize "792x664!" -quality 100 map-scaled.png
convert mk52.png \( map-scaled.png -alpha set -channel a -evaluate set 90% \) -geometry  +20+102 -composite blend.png

rm -f map.png map-scaled.png map-legend.png
```

[image]

The matplotlib docs include examples of image manipulation using Pillow, but I haven't looked into that yet.

Background image used:

[image]

VileBrigandier, 2018-10-24:
Much work has been done on this, closing as this issue is far out of date with the plugin's current form.

closed 2018-10-24 by VileBrigandier