PrusaOwners/OctoPrint-PrusaMeshMap #9: Enhancement request: add option to enable auto refresh of image

Patronics, 2018-04-02:
I think it would be useful to have an option to enable auto-refreshing of the heatmap as an option in settings or as a checkbox near the “refresh heatmap” button. Also if it kept a record of the previous (or past few) measurements, you could get the same effect as currently allowing you to compare “before” and “after” heatmaps, while still enabling a “live” view without needing to refresh. (Disclaimer: I have yet to install this plugin, as a long print is currently

Patronics, 2018-04-02:
... ongoing, so I don’t know how well this would work in practice.) (sorry for accidentally hitting submit too soon, leading to the split post)

ff8jake, 2018-04-02:
This is actually one of the more difficult things I ran into with this plugin, and I am still not certain it's optimal (but nobody in #octoprint answered my question when I asked it there, so I went with it for now).

When the heatmap.png file gets loaded, your browser caches it. To work around this, the "reload" button tells Javascript to reload it with heatmap.png?someDateTimeStamp. This fools your browser into reloading heatmap.png instead of using cache. The downside is now you have more than one heatmap.png cached at the browser end, and if you were to use this method in an autorefresh... you can see how it gets ugly fast. :)

I am considering storing multiple images by timestamp and having the plugin give the user the option to view past results. My issue currently is I don't know the proper way OctoPrint wants me to store them. Right now heatmap.png is served up out of site-packages/octoprint_PrusaMeshMap/static/img/ directory, and I think I shouldn't be overwriting it continually, and most certainly shouldn't keep storing data there. I couldn't find any documentation on where these should be stored or how to get to them from the public facing webserver either.

So marking this one as help wanted. I will try to look at it some more, but maybe someone with more experience than me in this area can contribute. :)