PrusaOwners/OctoPrint-PrusaMeshMap #23: Heatmap Image URL incorrect when using reverse proxy

ClassicGOD, 2018-06-21:
My Octoprint instance is behind a Nginx reverse proxy and the heatmap image does not load.  
Plugin generates incorrect url ignoring X-Script-Name for example for octoprint installation avaiable under https://myDomain.com/octoprint/ :

url of the image is: https://myDomain.com/plugin/PrusaMeshMap/static/img/heatmap.png

but it should be: https://myDomain.com/octoprint/plugin/PrusaMeshMap/static/img/heatmap.png

lord-carlos, 2018-07-01:
I have this problem as well.

lord-carlos, 2018-07-11:
For the time being I added this filter in Nginx, where XXXX is my subdirectoy.

>subs_filter_types text/javascript;
subs_filter 'plugin/PrusaMeshMap' 'XXXX/plugin/PrusaMeshMap' o;

lord-carlos, 2018-07-11:
For now I wrote a rule in nginx to rewrite the URL on the fly:

>subs_filter_types text/javascript;
subs_filter 'plugin/PrusaMeshMap' 'XXXX/plugin/PrusaMeshMap' g;

Where XXXX is my Subdirectory.

ppamidimarri, 2018-10-31:
I think this should fix the problem... #33