PrusaOwners/OctoPrint-PrusaMeshMap #24: Reload Heatmap Image

jt11racin, 2018-06-27:
I have tries many times to reload image. No response. Can not tell if the button works or is working on bringing it up. I prefer Firefox.

Arksine, 2018-06-27:
If you are using NoScript, Adblock, or UBlock you need to whitelist your octoprint instance.

lord-carlos, 2018-07-01:
And if you use a reverse proxy you need to map the /plugin directory.

david-mikulin, 2019-02-06:
Whitelisting the OctoPrint page in uBlock Origin worked for me.

tdub415, 2019-03-15:
Ah ha! I wondered why I couldn't ever refresh on my PC, just my phone.  Thanks, was driving me crazy!

Haprog, 2019-09-17:
Thanks! I also had "uBlock Origin" blocking it so that the image only showed up when I reloaded the whole page.

This should be mentioned in the readme or other documentation since it's a common pitfall.

Or this could probably be "fixed" by renaming the filename/URL for the image. I think the problem is that many ad blockers block resources which contain the word "heatmap". In my case it only seemed to get blocked when it was requested asynchronously (after pressing the button).