PrusaOwners/OctoPrint-PrusaMeshMap #35: Installation issues with Octopi

SurfingSpider, 2018-11-12:
Hi I am trying to install this plugin on my RPI Octopi and it fails. Can you offer any guidance? 

First it failed because PIP was out of date so I updated it with the command..

`/home/pi/oprint/bin/pip install --upgrade pip`

Then had also seen the note about updating matplotlib and numpy on the raspberry pi so ran 

` /home/pi/oprint/bin/pip --no-cache-dir install matplotlib numpy`
 
These both ran correctly but I get this install issue in the log file

```
2018-11-12 12:52:37,570   /home/pi/oprint/bin/python -m pip install https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap --no-cache-dir
2018-11-12 12:52:40,861 > Collecting https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap
2018-11-12 12:52:41,578 ! Cannot unpack file /tmp/pip-xqvXnF-unpack/OctoPrint-PrusaMeshMap (downloaded from /tmp/pip-IbiUjL-build, content-type: text/html; charset=utf-8); cannot detect archive format
2018-11-12 12:52:41,594 ! Cannot determine archive format of /tmp/pip-IbiUjL-build
2018-11-12 12:52:42,095 > Downloading https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap
2018-11-12 12:52:42,861 ! You are using pip version 8.1.1, however version 18.1 is available.
2018-11-12 12:52:42,862 ! You should consider upgrading via the 'pip install --upgrade pip' command.
2018-11-12 15:22:48,775   /home/pi/oprint/bin/python -m pip install https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap --no-cache-dir
2018-11-12 15:22:52,114 > Collecting https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap
2018-11-12 15:22:52,115 > Downloading https://github.com/PrusaOwners/OctoPrint-PrusaMeshMap
2018-11-12 15:22:52,617 ! Cannot unpack file /tmp/pip-unpack-3f5beY/OctoPrint-PrusaMeshMap (downloaded from /tmp/pip-req-build-NU5fUZ, content-type: text/html; charset=utf-8); cannot detect archive format
2018-11-12 15:22:52,617 ! Cannot determine archive format of /tmp/pip-req-build-NU5fUZ
```

SurfingSpider, 2018-11-12:
I have figured it out. I put the link to the ZIP file download into the URL section of the plugin manager. It works now

closed 2018-11-12 by SurfingSpider

Boby71, 2019-01-05:
Using the ZIP in the plugin manager doesn't work for me:
> Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-wLWGnN/matplotlib/
> You are using pip version 8.1.1, however version 18.1 is available.
> You should consider upgrading via the 'pip install --upgrade pip' command.
> Fehler!
> Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
Any ideas?