PrusaOwners/OctoPrint-PrusaMeshMap #36: Why is the plugin not showing up in the plugin manager?

Boby71, 2018-11-18:
The description says, I should use the built-in plugin manager - but I'm unable to find it there.
Running OptoPi on 1.3.9 and since yesterday 1.3.10 RC2. Something special I need to do to enable it?

fraschetti, 2019-01-12:
I ran into this same issue. For my particular case, here are the relevant OctoPrint log messages....

```
2019-01-12 16:19:57,477 - octoprint.cli.server - ERROR - Uncaught exception
ImportError: No module named _multiarray_umath
2019-01-12 16:19:57,478 - octoprint.plugin.core - ERROR - Error loading plugin PrusaMeshMap
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 847, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaMeshMap/__init__.py", line 16, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matplotlib/pyplot.py", line 31, in <module>
    import matplotlib.colorbar
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matplotlib/colorbar.py", line 32, in <module>
    import matplotlib.artist as martist
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matplotlib/artist.py", line 16, in <module>
    from .path import Path
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matplotlib/path.py", line 21, in <module>
    from . import _path, rcParams
ImportError: numpy.core.multiarray failed to import

```

Boby71, 2019-01-15:
My PIP installation was corrupt; I reinstalled OctoPi from the scratch and now I was able to install PrusaMeshMap via the link from Github (zip-master). But it still doesn’t show up in the plugin manager...

fraschetti, 2019-01-16:
I'd check your octoprint.log file ( ~/.octoprint/logs/octoprint.log for the OctoPi distributions) to see if it's reporting an error with the plugin.

To confirm my issue was specific to my local environment, I put a stock OctoPi image on a an SD card and ensured I installed mathplotlib and numpy via the provided command...
`/home/pi/oprint/bin/pip --no-cache-dir install matplotlib numpy`
That _fresh_ install approach didn't enoucnter any errors which confirmed the issue had to be with my setup.

The most likely (although not confirmed) cause for my local instance was a conflicting version of pip from my Python 3.7 installation. Although Python + Pip for OctoPrint should be isolated, one of my reinstall attempts which ran against the out of the box Python+Pip failed on the matplotlib compilation and referenced Python 3.7 directories. As best I can tell, installing Python 3.7 dropped a pip3 and **pip** exeutable in /usr/local/bin/ and that may have erroneously been used instead of the intended pip during parts of the compilation process.

This numpy issue referenced here is all over the web and each fix is obviously unique to each root cause.
For my particular issue where I had installed Python 3.7 and may not have specifically used the provided pip command to install matplotlib and numpy, here was my solution:

1. Delete /usr/local/bin/pip  (this executable isn't there in the stock OctoPi build and equivalent to the duplicate pip3 in my setup)
`sudo rm usr/local/bin/pip`
2. Delete the existing matplotlib and numpy builds (just in case)
`rm -rf /home/pi/oprint/local/lib/python2.7/site-packages/matplotlib*`
`rm -rf /home/pi/oprint/local/lib/python2.7/site-packages/numpy*`
3. And then install matplotlib and numpy per as suggested by the plugin devs
`pip install matplotlib numpy --no-cache-dir --upgrade --ignore-installed`

My situatation and solution are likely unique to my setup so I'd suggest folks take a look at octoprint.log first to assess the root cause and then go from there.

Good luck!

CoteRL, 2019-02-01:
It doesn't appear for me or even using the web version of the plugin repository. https://plugins.octoprint.org/ Searching for the plugin returns no results.

Arksine, 2019-02-01:
The plugin isn't part of Octoprint's repository, it has to be installed using the URL or archive method.  Unfortunately the original creator sold his MK3 and moved on, and at the moment no one is available to maintain it.  I'm doubtful that anyone will pick up development any time soon, if ever.

dovecode, 2019-03-09:
@Arksine, do you know if anyone has picked up this project? I notice that @mcm001's fork has more activity than this repo...

Arksine, 2019-03-09:
He was working on it for a time, but I believe he is busy with FRC.  I don't know the status of his fork (ie: how close it is to merging) or if he has any plans to continue development after the competition is complete.

This project's status remains dormant.  I'm leaving it open for the time being (along with all issues) in the event that someone decides to pick it up.  If its still inactive this by the summer I will likely archive it.

mcm001, 2019-03-09:
The fork is currently inactive due to my after school commitments eating
literally all my free time. I left the project in a state where it should
be workable but needed more testing. That being said I use it on my Klipper
mk3 and it's working well. I hope to wrap up this project in addition to
some other loose ends in a couple of months, so if you or anyone else would
like to contribute to the PR that would be greatly appreciated.

dovecode, 2019-03-09:
Is everything covered in issues on your fork, @mcm001? I don't mind taking a look, but I'm in no way a Python or gcode veteran...