PrusaOwners/OctoPrint-PrusaMeshMap #42: 7x7 Mesh freeze up

Noahghclark, 2019-03-14:
I am having a problem with the heat map freezing up and not changing.  I believe it is related to using the G80 N7 command in the modified firmware made by prusa fans which only adds this command to do 7x7 mesh leveling instead of the 3x3 mesh leveling.  The heat map works fine when doing the standard G80 command for 3x3.  However when using the G80 N7 command it will produce the heat map from the points and then not change again, even when using the standard G80 command.  I am able to correct the problem by uninstalling and reinstalling the plugin.  That is until I input the G80 N7 command again.  Can this be fixed somehow so I don't have to keep on uninstalling and reinstalling the plugin.

ThYr0N, 2019-04-03:
Same issue here, now that PRUSA oficially have released the 7x7, no image after the level test.

It's and awasome plugin please fix it :D

zawadzkip, 2019-04-05:
+1 for this.

bpvarsity, 2019-04-17:
Would love if someone could make this plugin work again. It's such a great tool for leveling the bed. I have the same issue of no image after the test. Doesn't matter if I do 3x3 or 7x7

bpvarsity, 2019-04-18:
To those reading this flash firmware 3.6.0 and the plugin works. Level the bed with 3x3 and the visualizer and then flash 3.7.0 to get 7x7 back. Unfortunately can't get the graph to work with 7x7 but 3x3 is better than nothing

JohnTRoth, 2019-05-08:
Open up /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrusaMeshMap/__init__.py and change lines 109 and 111:
MESH_NUM_MEASURED_POINTS_X = 7
MESH_NUM_MEASURED_POINTS_Y = 7

After doing that, it now generates an image that looks correct with 7x7 enabled.

bpvarsity, 2019-05-08:
Thanks will try that

ThYr0N, 2019-05-21:
Still no image, it does the measuring and then no image. Any way of reseting it?

bpvarsity, 2019-05-21:
I use this now and it works https://github.com/AndyQ/OctoPrint-PrusaMeshMap

jsodeman, 2019-12-17:
My problem with the image not reloading and showing was uBlock Origin was preventing it from loading. Adding the OctoPrint IP to the whitelist fixed it.