PrusaOwners/OctoPrint-PrusaMeshMap #43: Installation fails due to Matplotlib 3.x requiring python 3.x

r-a-i, 2019-04-07:
The new marplotlib needs python 3.  I encountered this while trying to install on a raspberry pi zero w. 

I worked around this by manually installing marplotlib 2.x `pip install ‘matplotlib<3.0’` before manually running the installer.

xoxota99, 2019-05-25:
I'm having the same problem, but I'm not even able to install matplotlib from the command line.

boraca, 2019-07-02:
You have to install older version of NumPy as well.

    /home/pi/oprint/bin/pip --no-cache-dir install 'numpy<1.17'
    /home/pi/oprint/bin/pip --no-cache-dir install 'matplotlib<3.0'

jmsaltzman, 2019-08-16:
Installing numpy and matplotlib on their own before the plugin worked for me. This should be on the front page with the installation instructions IMO.

cstrahan, 2019-08-31:
When installing numpy, I had to restrict the max number of concurrent jobs in order to install numpy successfully:

```
$ NPY_NUM_BUILD_JOBS=1 /home/pi/oprint/bin/pip --no-cache-dir install 'numpy<1.17'
```

otherwise I'd get a segfault like so:

```
    In file included from /usr/include/python2.7/pyport.h:325:0,
                     from /usr/include/python2.7/Python.h:58,
                     from numpy/core/src/umath/ufunc_type_resolution.c:15:
    /usr/include/arm-linux-gnueabihf/bits/mathcalls.h:115:1: note: expected 'long double *' but argument is of type 'npy_longdouble * {aka double *}'
     __MATHCALL (modf,, (_Mdouble_ __x, _Mdouble_ *__iptr)) __nonnull ((2));
     ^
    arm-linux-gnueabihf-gcc: build/src.linux-armv7l-2.7/numpy/core/src/multiarray/einsum.c
    In file included from numpy/core/include/numpy/npy_math.h:548:0,
                     from numpy/core/include/numpy/halffloat.h:5,
                     from numpy/core/src/multiarray/einsum.c.src:19:
    numpy/core/src/npymath/npy_math_internal.h.src: In function 'npy_modfl':
    numpy/core/src/npymath/npy_math_internal.h.src:490:21: warning: passing argument 2 of 'modfl' from incompatible pointer type [-Wincompatible-pointer-types]
         return modf@c@(x, iptr);
                         ^~~~
    In file included from /usr/include/python2.7/pyport.h:325:0,
                     from /usr/include/python2.7/Python.h:58,
                     from numpy/core/src/multiarray/einsum.c.src:12:
    /usr/include/arm-linux-gnueabihf/bits/mathcalls.h:115:1: note: expected 'long double *' but argument is of type 'npy_longdouble * {aka double *}'
     __MATHCALL (modf,, (_Mdouble_ __x, _Mdouble_ *__iptr)) __nonnull ((2));
     ^
    arm-linux-gnueabihf-gcc: numpy/core/src/multiarray/strfuncs.c
    arm-linux-gnueabihf-gcc: numpy/core/src/multiarray/number.c
    arm-linux-gnueabihf-gcc: build/src.linux-armv7l-2.7/numpy/core/src/multiarray/arraytypes.c
    arm-linux-gnueabihf-gcc: internal compiler error: Segmentation fault (program as)
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
    arm-linux-gnueabihf-gcc: build/src.linux-armv7l-2.7/numpy/core/src/umath/scalarmath.c
    error: Command "arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -std=c89 -Wno-error -fPIC -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -Ibuild/src.linux-armv7l-2.7/numpy/core/src/umath -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -Ibuild/src.linux-armv7l-2.7/numpy/core/src/common -Inumpy/core/include -Ibuild/src.linux-armv7l-2.7/numpy/core/include/numpy -Inumpy/core/src/common -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -I/usr/include/python2.7 -Ibuild/src.linux-armv7l-2.7/numpy/core/src/common -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -Ibuild/src.linux-armv7l-2.7/numpy/core/src/common -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -c numpy/core/src/multiarray/number.c -o build/temp.linux-armv7l-2.7/numpy/core/src/multiarray/number.o -MMD -MF build/temp.linux-armv7l-2.7/numpy/core/src/multiarray/number.o.d" failed with exit status 4
    In file included from numpy/core/include/numpy/npy_math.h:548:0,
                     from numpy/core/src/multiarray/arraytypes.c.src:16:
    numpy/core/src/npymath/npy_math_internal.h.src: In function 'npy_modfl':
    numpy/core/src/npymath/npy_math_internal.h.src:490:21: warning: passing argument 2 of 'modfl' from incompatible pointer type [-Wincompatible-pointer-types]
         return modf@c@(x, iptr);
                         ^~~~
    In file included from /usr/include/python2.7/pyport.h:325:0,
                     from /usr/include/python2.7/Python.h:58,
                     from numpy/core/src/multiarray/arraytypes.c.src:3:
    /usr/include/arm-linux-gnueabihf/bits/mathcalls.h:115:1: note: expected 'long double *' but argument is of type 'npy_longdouble * {aka double *}'
     __MATHCALL (modf,, (_Mdouble_ __x, _Mdouble_ *__iptr)) __nonnull ((2));
     ^
    In file included from numpy/core/include/numpy/npy_math.h:548:0,
                     from numpy/core/include/numpy/ufuncobject.h:4,
                     from numpy/core/src/umath/scalarmath.c.src:16:
    numpy/core/src/npymath/npy_math_internal.h.src: In function 'npy_modfl':
    numpy/core/src/npymath/npy_math_internal.h.src:490:21: warning: passing argument 2 of 'modfl' from incompatible pointer type [-Wincompatible-pointer-types]
         return modf@c@(x, iptr);
                         ^~~~
    In file included from /usr/include/python2.7/pyport.h:325:0,
                     from /usr/include/python2.7/Python.h:58,
                     from numpy/core/src/umath/scalarmath.c.src:13:
    /usr/include/arm-linux-gnueabihf/bits/mathcalls.h:115:1: note: expected 'long double *' but argument is of type 'npy_longdouble * {aka double *}'
     __MATHCALL (modf,, (_Mdouble_ __x, _Mdouble_ *__iptr)) __nonnull ((2));
     ^
    numpy/core/src/multiarray/arraytypes.c.src: In function 'small_correlate':
    numpy/core/src/multiarray/arraytypes.c.src:4919:1: internal compiler error: in estimate_operator_cost, at tree-inline.c:3959
     }
     ^
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
```

I'm guessing the device runs low on memory during compilation.

Haprog, 2019-09-17:
I just setup OctoPi (my first time using OctoPrint) on a new Raspberry Pi 4 B (with 4GB RAM) and wanted to install this plugin but got the same problem.

Thanks @boraca! With those install commands I was able to get this installed and working (though now I have another issue with the heatmap image not showing without reloading the whole page, I'll see if there's another issue about that open).